Ultimaker/Cura #17580: Failed Slicer

kevshiro, 2023-12-10:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Anycubic Kobra Max

### Name abnormal settings

Typically I use the Tree method for supports

### Describe model location

Doesn't seem to slice if I rotate it along z,x, or y

### Describe your model

IT sliced when it was smaller.  At about 1000 percent.  When I made it larger, it seemed to have issues slicing.

### Add your .zip here ⬇️

[AKMAX_Rainmaker.zip](https://github.com/Ultimaker/Cura/files/13626007/AKMAX_Rainmaker.zip)

5axes, 2023-12-10:
An other Tree support Issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic