Ultimaker/Cura #17581: Slicing preview shows entire model mid-level

botolo, 2023-12-10:
### Cura Version

5.5.0 and 5.6.0

### Operating System

MacOS Sonoma

### Printer

Ender 3 V3 SE and Custom Settings for K1

### Reproduction steps

1. I opened a model I created on Tinkercad
2. I sliced the model
3. Layer 486 shows the entire model in preview (that layer is roughly at 50/60% of the model.

I am attaching a video that shows the issue.

https://github.com/Ultimaker/Cura/assets/3123816/168211c5-be31-4947-bb70-77afc6bbddb1

### Actual results

See above

### Expected results

See above

### Add your .zip and screenshots here ⬇️

https://github.com/Ultimaker/Cura/assets/3123816/b544fe19-ba4e-42a2-aec6-fe0095f51a74

GregValiant, 2023-12-10:
I get this occasionally when I'm using One-at-a-Time mode.  Did you happen to have that checked?  (It might help to know to narrow this down.)

botolo, 2023-12-12:
I just checked this, and the settings is "all at once". Here is the STL file that is generating this issue (this is something I designed on Tinkercad). FYI, I am not getting this issue when slicing this file with PrusaSlicer.

[Ciotola V2 Taller.stl.zip](https://github.com/Ultimaker/Cura/files/13642190/Ciotola.V2.Taller.stl.zip)

GregValiant, 2023-12-12:
The model has a shelf at layer 486.
Here I've laid the model down.  You can see that the top piece sits on a point on the bottom cylinder.  My guess would be that the width of that point is approaching zero and it is causing an inconsistent slice.  You can see in the X-ray view that there is a ledge all the way around.
[image]

There is also a 0.02mm diameter hole in the top.  Printing may make it larger, or it may disappear.
[image]

If you set the model to Z= -97 and slice it you will see that there is an empty layer there.  This model will print as two separate pieces.

So are those design features or flaws?

GregValiant, 2026-10-01:
I'll go ahead and close this.

closed 2026-10-01 by GregValiant