Ultimaker/Cura #17582: Screw thread slicer failure

Lord-RoadRunner, 2023-12-10:
### Cura Version

5.6.0

### Operating System

Win 10 Pro , Ryzen 2600 RTX 3060 driver version 546.29

### Printer

Ender 3 v2

### Name abnormal settings

no, default esun pla+ Grey

### Describe model location

Center, rotating the model 90dec and back allowed it to slice 

### Describe your model

no first attempt at sliceing this model, the screw and thread are connected as a single piece 

### Add your .zip here ⬇️

[CE3E3V2_annoying_box_thread_test.zip](https://github.com/Ultimaker/Cura/files/13626431/CE3E3V2_annoying_box_thread_test.zip)
 to this section

5axes, 2023-12-10:
So many Solution to avoid issue.. 
A. Don't need support. 
B. Change Part Orientation
C. Change the distance between models to allow Tree Support creation.

[image]

Anyway a Duplication of the Tree support Issue. It's very similar to this case : https://github.com/Ultimaker/Cura/issues/17529

Clearly @GregValiant  it's more and more boring all this report withour recaction of the UltiMaker Team.

GregValiant, 2023-12-10:
Yeah.  I'm just cherry picking.
As a matter of fact, I've had about enough.
Goodbye to all and a Merry Christmas.

Lord-RoadRunner, 2023-12-14:
I resolved the issue myself. The error I received was a unknown error so I wasn't sure how to solve the issue but I flipped the model 90° one way then back to original and this allowed it to properly slice

closed 2023-12-14 by Lord-RoadRunner