Ultimaker/Cura #17585: Cura Slicer crashes while attempting to slice

zumaster20, 2023-12-10:
### Cura Version

5.6.0, 5.4.0

### Operating System

windows 10

### Printer

Ender 3

### Name abnormal settings

no, model doesn't slice anyway

### Describe model location

model placed at the center of buildplate

### Describe your model

described in comment section

### Add your .zip here ⬇️

weird part is the fact, that previously models were slicing normally with like 30% chance of failure (crash like now), then every second attempt to slice began to cause crash, now I can not slice it at all. I checked logs, and error is  

[35m2023-12-10 14:31:15,834[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.834] [info] Progress: support accomplished in 0.013s[0m
[35m2023-12-10 14:31:15,834[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.834] [info] Starting export...[0m
[35m2023-12-10 14:31:15,850[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.851] [info] Progress: export accomplished in 0.016s[0m
[35m2023-12-10 14:31:15,851[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.851] [info] Sending 21 layers.[0m
[35m2023-12-10 14:31:15,851[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.851] [info] Total time elapsed 0.033s[0m
[35m2023-12-10 14:31:15,851[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [0m
[35m2023-12-10 14:31:15,852[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.851] [info] Print time (s): 321[0m
[35m2023-12-10 14:31:15,852[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.851] [info] Print time (hr|min|s): 0h 5m 21s[0m
[35m2023-12-10 14:31:15,852[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:15.851] [info] Filament (mm^3): 198[0m

"[35m2023-12-10 14:31:16,114[0m - [36mDEBUG[0m - [37m[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-12-10 14:31:16.115] [info] Closing connection.[0m
 [35m2023-12-10 14:31:16,326[0m - [33mWARNING[0m - [37m[MainThread] UM.Backend.Backend._onSocketError [267]: Unhandled socket error ErrorCode.ReceiveFailedError[0m"
 
 
i think its last 4 lines
 everything before seems pretty right to me, Im getting print time, mass of printed model and all this stuff, and then this error occurs, and cura just stop responding
 
 
[selector_project_issue.zip](https://github.com/Ultimaker/Cura/files/13627536/selector_project_issue.zip)

zumaster20, 2023-12-10:
also everything works fine if Im doing it not on my pc, but in my laptop
pc:
gpu AMD Radeon RX5700XT
cpu AMD Ryzen 7 3600
mb ASRock SteelLegend B450M
16gb ddr4 RAM
os Win10

laptop: 
cpu Intel I5-4200
os Win10

5axes, 2023-12-10:
No issue for me 
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic