Ultimaker/Cura #17590: Slicing failed

SargentK, 2023-12-10:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Kobra Plus

### Reproduction steps

Opened a model that I purchased and tried slicing.

### Actual results

Slicing Failed with an unexpected error. Please consider reporting a bug on our issue tracker

### Expected results

I should have been able to slice and print

### Add your .zip and screenshots here ⬇️

Not sure what other info that you want other than I continue to have more trouble printing with Cura than I do with other programs

5axes, 2023-12-12:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

GregValiant, 2026-10-01:
5.4, 5.5 and 5.6 had problems with certain types of models.  Most of that was fixed in 5.7.0 and since then, there have been numerous improvements.  The current version is 5.13.0 so it is time to update.

closed 2026-10-01 by GregValiant