Ultimaker/Cura #17591: My first error message.

Iyakimaru, 2023-12-10:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3-s1

### Describe your model

I just downloaded it from https://www.thingiverse.com/thing:5574662 It is a textured roll. It's the first time I use it.

### Add your .zip here ⬇️

[CE3S1_wood_roller.zip](https://github.com/Ultimaker/Cura/files/13628621/CE3S1_wood_roller.zip)

5axes, 2023-12-11:
Not sure you needs some support for this part. AnyWay an other Tree Support Issue.
https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0