Ultimaker/Cura #17592: Unidentified slicing error

Dinohunterandmuncher, 2023-12-10:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

irrelevant

### Reproduction steps

1. export odd shape.
2. slice

### Actual results

Cura can't identify the fact it didn't slice right, now I'm left with areas where there is no print. I believe this affects the print as every time I tried this print it kept printing nothing then a mess.

### Expected results

Everything should be there, the base plate and supports should also be there, nothing was there except half the print. so my printer only did half the job.

### Add your .zip and screenshots here ⬇️

Within this zip file I have the following: the blender file which I build odd shape with, STL which I make thing print with, and screenshots of what I have been getting.
[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/13628725/New.Compressed.zipped.Folder.zip)

5axes, 2023-12-10:
Model invalid 
[image]

Dinohunterandmuncher, 2023-12-12:
I... I guess it was my fault the print didn't work then, well, I'm sorry to bother you then, I guess it makes sense due to the fact it worked on a unaltered cube.

closed 2023-12-12 by Dinohunterandmuncher