Ultimaker/Cura #17594: [5.5.0 and 5.6.0] First Layer Wallspeed too fast

CPT-Firefly, 2023-12-10:
### Cura Version

5.5.0 and 5.6.0

### Operating System

Windows 11

### Printer

Creality CR-5 Pro H with Marlin

### Reproduction steps

I almost always print the first layer at 10 mm/s. I cannot use Cura 5.5.0 and 5.6.0 because the first layer ignores this filament profile setting. I tested different filament profiles and the problem remains. This means I can currently only use Cura 5.4.0.

### Actual results

No change.

### Expected results

No change.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[COUVERCLE PERFO Cura 5.4.0.zip](https://github.com/Ultimaker/Cura/files/13628845/COUVERCLE.PERFO.Cura.5.4.0.zip)
[COUVERCLE PERFO Cura 5.6.0.zip](https://github.com/Ultimaker/Cura/files/13628846/COUVERCLE.PERFO.Cura.5.6.0.zip)

GregValiant, 2023-12-11:
Looks like a bug to me.  Only the skins seem to be affected by the low initial layer speed.

gfnord, 2023-12-30:
I see the same issue here, wall speed on the first layers does not use the Initial Layer Print Speed, it uses the wall speed and this causes the filament to not stick to the bed.

011V32, 2024-01-02:
I have the same issue with 5.5 and 5.6.

GregValiant, 2024-01-02:
I have looked into this further.
The Speed setting "Flow Equalization Ratio" appears to be the culprit.  When that setting was first implemented it would slow down the print head if you had a heavy flow.  An example would be putting down the infill at 2X layer height to speed up a print.
Something was changed because now what I'm seeing in the gcode is Flow Equalization for thin lines (low flow rate through the nozzle) is causing print speed increases.  Cura is adjusting the speed upward to keep the flow rate steady.  I'm currently running it at 50% and the situation is much improved.  To eliminate the situation all together - set the Flow Equalization Ratio to 0%.  

This is a chart of print speeds for a Benchy.
[image]

You can see that there are some speeds over 100mm/sec.  The smoke funnel is printed last and it's slow because those layers are bouncing off the Minimum Layer Time.
The "situation" occurs when the model has a wall that isn't an exact multiple of the line width.  The narrow lines in the center of the wall are printed faster and sometime, a lot faster that what the settings are.

In my opinion this isn't really a bug.  The code is doing what it was designed to do.
I just don't like the design.

CPT-Firefly, 2024-01-02:
Thank you GregValiant, 
I've been using this PETG profile since about 5.0. Up to and including 5.4. the first layer including the walls was printed at the speed for the first layer.  Line width of the first layer is 0.48 (120% of 0.4 line width).  The Flow Equalization Ratio is 100% as standard, I have not changed this value in any of my profiles.  Since 5.5, the walls of the first layer are not printed at the speed for the first layer but at the wall speed.

gfnord, 2024-01-02:
> Flow Equalization

Hi Greg, 

I changed the Flow Equalization to 0% and still it did not make the first wall layer print at the slow speed of first layer, thus affecting the quality of the printing. 

Yes, the code always does what it is programmed to do. In my point of view, the Initial Layer Print Speed should take precedence over all other settings, since it is very important for the adhesion.

I had to go back to version 5.4.

Kathode-Ray, 2024-01-02:
This looks a lot like this issue:

[Cura 5.5 first layer perimeter speed inconsistent #17210](https://github.com/Ultimaker/Cura/issues/17210)

GregValiant, 2024-01-02:
Yes, this issue is different.  There is another report #17858 that is saying that the first layer acceleration is also wrong.  I had no problem duplicating that report.  This one and that one may be related.

5axes, 2024-01-04:
Temporary solution : https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/CheckFirstSpeed.py

joelmdev, 2024-01-18:
Changing Flow Equalization to 0% had no effect for me, however as suggested in the Ultimaker forums enabling "Print Thin Walls" did alleviate the issue. This is an acceptable workaround for models with no thin walls to speak of as it has no ill effect on slicing.

Flight, 2024-02-10:
Still an issue on my Sovol SV06. Tried to play with the flow equalisation and the Print thin walls, both doesn't help. It still prints my first layers at the half of the maximum speed, instead of the First layer speed.

davispw, 2024-03-22:
I'm experiencing this issue with 5.6.0 on some models, with and without Flow Equalization Ratio = 0%, with and without Print Thin Walls.

davispw, 2024-03-22:
I don't see the problem with 5.7.0.beta.1, same model and settings.

GregValiant, 2026-10-01:
I'll close this as it seems to have been fixed.

closed 2026-10-01 by GregValiant