Ultimaker/Cura #17595: Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

discou, 2023-12-10:
### Cura Version

5.6.0

### Operating System

Windows 11 - 22H2

### Printer

Snapmaker 2 A350

### Reproduction steps

load an stl, set TPE filament from XYZPrinting and change some parameters : 30% infill, printspeed at 20mm/s, retraction off, wth fast profile.  Then I tried to slice it.

### Actual results

An error occured : "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker."

### Expected results

Having a gcode file...

### Add your .zip and screenshots here ⬇️

[image]

[S2A_Plug vis pour sapin v3_Cura.zip](https://github.com/Ultimaker/Cura/files/13629131/S2A_Plug.vis.pour.sapin.v3_Cura.zip)

5axes, 2023-12-12:
No issue on Windows 10 
[image]

An other Windows 11 issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

GregValiant, 2026-10-01:
5.5 and 5.6 had problems with certain types of models.  Most of that was fixed in 5.7.0 and since then, there have been numerous improvements.  The current version is 5.13.0 so it is time to update.

closed 2026-10-01 by GregValiant