Ultimaker/Cura #17597: Slicing failed due to tree support structure - fire pipe clip

Beep6581, 2023-12-10:
### Cura Version

5.6.0

### Operating System

Manjaro

### Printer

PrimaCreator P120 (Malyan M200)

### Name abnormal settings

The issue goes away if I change support structure from "tree" to "normal"

### Describe model location

Centered. Moving makes no difference.

### Describe your model

Have not sliced it before. It is watertight. Have tried mesh fixes - no difference, still fail on tree support.

### Add your .zip here ⬇️

[MM_fire pipe clips.zip](https://github.com/Ultimaker/Cura/files/13629293/MM_fire.pipe.clips.zip)

5axes, 2023-12-10:
No issue on Windows 10

[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0