Ultimaker/Cura #17598: Slicing failed on mirror image of a piece I just printed.

DowntownTrainer, 2023-12-10:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Custom CR10 profile

### Name abnormal settings

First time getting a slice fail, using same settings as I normally use

### Describe model location

Still won't slice

### Describe your model

Just downloaded it from Thingiverse.com thing:6306571 part 2 of 2 file

### Add your .zip here ⬇️

The other side printed fine, this is a mirror image of an angel wing christmas ornament.  First time getting a slicer failure
[Wing_Ornament_-_Part_2_of_2a.zip](https://github.com/Ultimaker/Cura/files/13629347/Wing_Ornament_-_Part_2_of_2a.zip)

5axes, 2023-12-11:
Rotation of the part of 90° should temporary solved your issue.
[image]

DowntownTrainer, 2023-12-11:
I'm thinking we can close this issue, the actual file, part 2 of 2 seems to have incongruity issues.  I just mirror imaged part 1 of 2 and it sliced fine.

closed 2023-12-11 by DowntownTrainer