Ultimaker/Cura #17600: Slicing failed

Bernardino-Home, 2023-12-10:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3

### Name abnormal settings

Didn't changed setting

### Describe model location

Doesn't slice in any rotation

### Describe your model

First time slicing this model

### Add your .zip here ⬇️

[upper-body.zip](https://github.com/Ultimaker/Cura/files/13629537/upper-body.zip)

5axes, 2023-12-11:
Doesn't have any issue with this model.
[image]

Don't share an STL file it's absolutly useless, share your project file for analyse.

Bernardino-Home, 2023-12-11:
[image]

Close bug.
Today it started to work.

The only difference from today and yesterday was that today, i had the pen inserted on the slot.

Bernardino-Home, 2023-12-11:
Today's working fine. the only difference was me having the pen inserted. As you can see, yesterday I had this problem:
[image]

But not anymore, don't know why.

Bernardino-Home, 2023-12-11:
Thank you and i'm sorry for the trouble:

[image]

closed 2023-12-11 by Bernardino-Home