Ultimaker/Cura #17605: Support problem

Juraj-MJ, 2023-12-11:
### Cura Version

5.6.0

### Operating System

Windows 10 

### Printer

Ultimaker S5

### Describe your model

mesh fix doesnt work

### Add your .zip here ⬇️

[Objekt T.zip](https://github.com/Ultimaker/Cura/files/13633693/Objekt.T.zip)

5axes, 2023-12-12:
STL are useless for analyse and Bug Report .. You must share your Cura Project.
No issue to slice : 
[image]

Juraj-MJ, 2023-12-12:
start tree support, and size 200%

5axes, 2023-12-12:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

Juraj-MJ, 2023-12-12:
i cant attach file, becouse in zip it has 31 mb ...

5axes, 2023-12-12:
Certainly an other Tree Support issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Juraj-MJ, 2023-12-13:
https://wetransfer.com/downloads/1a4fb15b17b70176cd3956f86059723220231213092538/177ad7f05d9119bcd548d26b63d6c8df20231213092615/5c979c  

try this, i insert there Cura project and print sceen

5axes, 2023-12-13:
Not a Cura Project just the 3mf format ... (No Printer and setting information) ...  But don't waste your time, as it is unlikely that the development team will be interested in your problem, current data is sufficient.