Ultimaker/Cura #17606: Slicing failed

apkelly, 2023-12-11:
### Cura Version

5.5.0

### Operating System

Mac OS Ventura

### Printer

Flsun Q5

### Describe your model

Sliced at draft setting 0.2mm, failed at fine 0.1mm

### Add your .zip here ⬇️

[FQ5_singleWallKomplett.3mf.zip](https://github.com/Ultimaker/Cura/files/13634587/FQ5_singleWallKomplett.3mf.zip)

5axes, 2023-12-11:
No issue for me :

[image]

apkelly, 2023-12-11:
That isn't my model

5axes, 2023-12-12:
Oup's ... 
[image]

An other **Print Thin Wall** Issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

MariMakes, 2024-01-18:
Hi @apkelly,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0