Ultimaker/Cura #17613: Slicing failed

bsprus, 2023-12-12:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Custom FF (Kobra 2 Max)

### Name abnormal settings

support problem

### Describe model location

no

### Describe your model

overhang test

### Add your .zip here ⬇️

[Fail.zip](https://github.com/Ultimaker/Cura/files/13642236/Fail.zip)

5axes, 2023-12-12:
No issue on Windows 10 ...

[image]

an other Windows 11 issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic