Ultimaker/Cura #17615: problems slicing this is the 3rd time i've had this issue

Jezza76, 2023-12-12:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

ender 3 v2

### Describe model location

its a large model fits across the build plate

### Describe your model

Gambody model should work fine from memory I have sliced before

### Add your .zip here ⬇️

[Cura Settings.pdf](https://github.com/Ultimaker/Cura/files/13646303/Cura.Settings.pdf)
[Uploading 796-11-11_body_front_FDM_(repaired).zip…]()

Jezza76, 2023-12-12:
Sorry sound a bit off in the heading lol sames to fail slice my be due to slicing while I'm printing using the monitor USB print function ill have a look and get back when I've finished my print shouldn't be long only a test print

5axes, 2023-12-12:
No Zip File attached to this issue