Ultimaker/Cura #17623: [5.6.0] Minor typo.

Asterchades, 2023-12-12:
### Cura Version

5.6.0

### Operating System

Windows 11 (23H2)

### Printer

Custom FFF Printer (default config)

### Reproduction steps

1. Download the file "Part_Studio_1_-_Hide.3mf from [this Thingiverse object](https://www.thingiverse.com/thing:6225612),
2. Attempt to open the file in Cura

Please note that this issue is not regarding the model itself. It appears to be a deliberately non-conformant 3MF exported by Bambu Studio.

### Actual results

A warning is presented stating that "There _where_ no models in" the file (emphasis mine).

### Expected results

"Were" is the appropriate spelling in this situation, being the past tense of "are" (which would also be acceptable).

### Add your .zip and screenshots here ⬇️

[image]

5axes, 2023-12-12:
Duplicate : https://github.com/Ultimaker/Cura/issues/16810

Asterchades, 2023-12-13:
My bad. Not quite sure how I missed those - they're recent.

closed 2023-12-13 by Asterchades