Ultimaker/Cura #17624: Christmas bells 3pk

kmcpickl, 2023-12-12:
### Cura Version

5.1.0

### Operating System

Windows 10 Home

### Printer

Creality Ender 3

### Name abnormal settings

none

### Describe model location

I moved it from the center and it sliced.

### Describe your model

Lattice Christmas bells that I have printed before

### Add your .zip here ⬇️

I have printed it before. I have opened the file with Elegoo Max 4 with the dedicated cura slicer for that model printer.
[CE3_open_lattice_bell 3pk compressed.zip](https://github.com/Ultimaker/Cura/files/13652915/CE3_open_lattice_bell.3pk.compressed.zip)

5axes, 2023-12-13:
No issue for me on Cura 5.6.0
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic