Ultimaker/Cura #17627: Cura 5.6 has problems with Anycubic Kobra Max

Rgray47, 2023-12-12:
### Cura Version

5.4.0 some problems / 5.6.0 real problems Z axis 

### Operating System

Windows 10

### Printer

Anycubic Kobra Max

### Reproduction steps

Sliced a previous model with the new version 5.6.0 and tried to print it.
[image]

### Actual results

It seems to forget the Z axis height after printing the first few layers.  The print head keeps crashing into the previous printed layers. 

### Expected results

Should have a normal 3D print

### Add your .zip and screenshots here ⬇️

Last known good version 5.2.1
[KobraMaxCura5.2.1_Sample_card_v1.0_ABS.zip](https://github.com/Ultimaker/Cura/files/13652104/KobraMaxCura5.2.1_Sample_card_v1.0_ABS.zip)
V5.6.0
[KobraMaxCura5.6.0_sample_card_v1.0_ABS.zip](https://github.com/Ultimaker/Cura/files/13652387/KobraMaxCura5.6.0_sample_card_v1.0_ABS.zip)

GregValiant, 2023-12-12:
Thanks for the report.
I commented on this on the Ultimaker Forum and it is as I thought.
Both files show that Z-hops were enabled at 0.075 height and both files show the Z moving up.  There are no dips back down into the model.  Throughout the file are the Z-hop pairs like this:
G1 F1200 **Z1.675** >>> Hop up
G0 F6000 X176.674 Y205.623 **Z1.675** >>> Travel at Z hop height
G0 X201 Y201.312
M204 S1000
M205 X8 Y8
;TYPE:WALL-INNER
G1 F1200 **Z1.6** >>> drop back down to the working height of the layer.
The 5.6 file has fewer Z-hops than the 5.2.1 file because the "combing" mode is different between the files.  Whatever is causing the "diving into the print" problem is not in the gcode.  The printer is being given the correct information, but isn't following it.

I'll go ahead and remove the bug label and close this as it isn't a Cura problem.  I hope you understand.
As I said, if this was a Cura problem there would be thousands of reports here and no one would use the software.

I would suggest you check things on the printer.  If the set screw that holds the Z threaded rod into the motor coupler comes loose then the Z can become unpredictable.  Loose trolley wheels, a loose print head or hot end.  There are mechanical things to look at.  The problem simply isn't in the gcode.

Rgray47, 2023-12-12:
Greg,

The printer works as expected.  The first thing I did was check with the Owl g code print.  Printed just fine.

I have gone back and forth with Cura V 5.6.0 and V 5.2.1, all prints sliced with Cura 5.2.1 is successful.

Still working on the actual g codes given from both versions to check for a difference.

Thanks,

Robert Gray
Senior Electrical Eng.
ITC
602-791-2074

GregValiant, 2023-12-13:
@Rgray47 are you printing via the USB card or some other way?

Rgray47, 2023-12-13:
USB to a SD card into the 3D Printer.  All prints are done this way.

Thanks,

Robert Gray
Senior Electrical Eng.
ITC
602-791-2074

GregValiant, 2023-12-13:
This is a chart of the Z moves in the gcode I generated with 5.6.0 using your project file.  Everything is moving up.  There are a lot of z moves because "Z-hops" were enabled.
[image]

Rgray47, 2023-12-13:
> Here ya go. This is direct from the 5.6.0 project file you posted above.
> Gcode file.zip<https://github.com/Ultimaker/Cura/files/13665010/Gcode.file.zip>
>
> -

Greg,

This Gcode gave me a perfect print.

So the question is, why did my original version of 5.6.0 gave me a bad print?

Can you send me a good link to Cura 5.6.0 so I can down load that one?

Thanks,

Robert Gray
Senior Electrical Eng.
ITC
602-791-2074

GregValiant, 2023-12-13:
It's just a couple of doors over at [Cura Releases](https://github.com/Ultimaker/Cura/releases).  Scroll down below the 5.6 advertising blurbs and you will find the download links.