Ultimaker/Cura #17630: too many travels

Taxom, 2023-12-12:
### Cura Version

5.5.0

### Operating System

W10x64

### Printer

Marlin

### Reproduction steps

open .stl
slicing 

### Actual results

too many travels
[image]

### Expected results

Less empty movements

### Add your .zip and screenshots here ⬇️

[UPS.zip](https://github.com/Ultimaker/Cura/files/13654179/UPS.zip)
[image]

https://github.com/Ultimaker/Cura/assets/20313019/139675c9-9173-4acc-9972-168217695235

Asterchades, 2023-12-13:
There are two factors at play here.

First appears to be a `Combing` issue. I believe this is related to #14511, and it looks to be exacerbated by the use of `Avoid Printed Parts When Traveling` creating some very long (and frankly erroneous) travel paths. This can be mostly cleaned up to varying degrees by using `Not in Skin`, `All`, or `None`, or by disabling `Avoid Printed Parts When Traveling`.

Second is the use of a `Top Surface Skin Layer`. Even though the properties appear to be set identically, this causes the very top layer/s of any skin to be treated as a separate feature to any other skin. Particularly with the lettering this can result in a lot of additional travels to both avoid this feature and return to it later, and `Combing` (combined with `Avoid Printed Parts When Traveling`) makes things worse again. As you're not using different properties for these layers, setting the `Top Surface Skin Layers` to zero (0) should not only have no appreciable impact on print quality but removes _all_ of the errant travel movements - even without changing the `Combing` setting.

5axes, 2023-12-13:
**Top Surface Skin Layers  = 1**
[image]

**Top Surface Skin Layers  = 0**
[image]

Taxom, 2023-12-13:
I often use Top Surface Skin Layers of Concentric.
I tried to change the Combing mode but it does not help.
But this is definitely a problem with Top Surface Skin Layers. Since the installation of parameter in 2 layers transfers the problem to the layer below. Is there any hope that this problem will be solved in the following updates? Since I often encounter this problem, even when there are not many details on the upper layer. Sorry for my English.