Ultimaker/Cura #17635: Slicing failed Bug after software update

ftm-raghav, 2023-12-13:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

I added the file. 

Set my paraments and tried to slice it. 

A bug error pop up message shows up as it fails slicing.

### Actual results

I see a Github link to report bug.

### Expected results

I expect the issue resolved and files being sliced to work smoothly

### Add your .zip and screenshots here ⬇️

[UMS5_Brick_set_Inside.zip](https://github.com/Ultimaker/Cura/files/13664158/UMS5_Brick_set_Inside.zip)

5axes, 2023-12-13:
No issue on Windows 10 . An other Windows 11 issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

MariMakes, 2024-01-08:
Hey @ftm-raghav,

Welcome to the UltiMaker Cura Github 🚀
Sorry, it took us a while to get back to you 😞

I'm on Windows 11 and I took a look at it in Cura 5.6 and I've been able to slice it, so I don't believe that it's a windows 11 issue but something else related to your set-up 🤔

[image]

I'll bring it up to the team to see what they can do to improve it. Fingers crossed 🤞 

Thanks for your report 👍

closed 2024-04-03 by HellAholic