Ultimaker/Cura #17636: Slicing Failed

Fubinian, 2023-12-13:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Ultimaker S5

### Describe model location

Yes-Moved about 1 inch to the left

### Describe your model

Yes- but it was only about 85% of the size that failed

### Add your .zip here ⬇️

File requested: [UMS5_The Web_Stairs.zip](https://github.com/Ultimaker/Cura/files/13664871/UMS5_The.Web_Stairs.zip)
.stl file that caused the slicing to crash: [The Web_Stairs.zip](https://github.com/Ultimaker/Cura/files/13664873/The.Web_Stairs.zip)

Please let me know if you are unable to access the file. I was able to successfully slice the file upon moving my model from the center of the build plate.

5axes, 2023-12-14:
An other Windows 11 issue :  https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452
[image]

Duplicate :  Slice Failed on Windows 11

https://github.com/Ultimaker/Cura/issues/17635
https://github.com/Ultimaker/Cura/issues/17617
https://github.com/Ultimaker/Cura/issues/17613
https://github.com/Ultimaker/Cura/issues/17612
https://github.com/Ultimaker/Cura/issues/17595
https://github.com/Ultimaker/Cura/issues/17571
https://github.com/Ultimaker/Cura/issues/17567
https://github.com/Ultimaker/Cura/issues/17566
https://github.com/Ultimaker/Cura/issues/17552
https://github.com/Ultimaker/Cura/issues/17546
https://github.com/Ultimaker/Cura/issues/17538
https://github.com/Ultimaker/Cura/issues/17530
https://github.com/Ultimaker/Cura/issues/17526
https://github.com/Ultimaker/Cura/issues/17523
https://github.com/Ultimaker/Cura/issues/17509
https://github.com/Ultimaker/Cura/issues/17491
https://github.com/Ultimaker/Cura/issues/17490
https://github.com/Ultimaker/Cura/issues/17461 ??
https://github.com/Ultimaker/Cura/issues/17380
https://github.com/Ultimaker/Cura/issues/17343
https://github.com/Ultimaker/Cura/issues/17335
https://github.com/Ultimaker/Cura/issues/17253

Recurring cases of slicing issues on Windows 11 which, when tested on Windows 10 workstations, generate no error.

GregValiant, 2023-12-26:
#17775 is another Windows 11 problem.

closed 2024-04-03 by HellAholic