Ultimaker/Cura #17637: Slicing button does not work

JambonRebel, 2023-12-13:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Snapmaker J1

### Name abnormal settings

Does not slice with default settings

### Describe model location

No

### Describe your model

Table

### Add your .zip here ⬇️

[4xIngredient_table_2_tubes.zip](https://github.com/Ultimaker/Cura/files/13666524/4xIngredient_table_2_tubes.zip)

I rotate the table so the top is facing down. No support should be required. I tried with multiple files and the slicing button is unclickable. It doesn't crash. It just wouldn't even start to slice

5axes, 2023-12-14:
No issue for me... You must share a Cura Project for analyse.

[image]

GregValiant, 2026-10-25:
I'll go ahead and close this.  Much was fixed in Cura 5.7

closed 2026-10-25 by GregValiant