Ultimaker/Cura #17638: slicing failed

rhntr64, 2023-12-14:
### Cura Version

5.4

### Operating System

windows 11

### Printer

Flsun Super Racer

### Name abnormal settings

Exclusive

### Describe model location

no

### Describe your model

Tail part for airplane

### Add your .zip here ⬇️

[FSR_horizontal stabilizer left.zip](https://github.com/Ultimaker/Cura/files/13668003/FSR_horizontal.stabilizer.left.zip)

5axes, 2023-12-14:
Could be due to the invalid Model :  
[image]

rhntr64, 2023-12-15:
Wow. That is strange.  That is not on the model when sliced with another slicer.  The part is from the Pond Racer 1200mm on Cults3d. He has internal structure requiring the use of the exclusive setting to have it populate. This model has been extensively downloaded and printed with limited isssues.Sent from my iPhone

image.png (view on web)

closed 2024-04-03 by HellAholic