Ultimaker/Cura #17639: Snapmaker J1 add on

SnapJ1, 2023-12-14:
### Cura Version

5.3.0

### Operating System

Win 11

### Printer

Snapmaker J1

### Reproduction steps

An object, "a vase", diam 200, height 200, base appr 130, won't "fit" to a working space 300x200x200.
Y should be reduced to 195.

An other object, like cylinder... z 200, y 197.... y should be reduced to 196 and z to 199.

### Actual results

Reduce size

### Expected results

Should fit...
(Fits with snapmakers own slicer, luban)

### Add your .zip and screenshots here ⬇️

Size in question

Asterchades, 2023-12-14:
The only time Cura will adjust the size of a given file is if it detects that it's been saved with the wrong unit of measurement - typically being saved in metres, rather than millimetres (which results in the model being only 1/1,000 the intended size). Granted it doesn't always detect these files correctly, and sometimes it detects files incorrectly, but that's the only situation it makes an adjustment should you have the setting enabled (which it is by default).

It's not going to scale a model down just because it doesn't fit your build volume. Doing so could easily result in a large amount of wasted time and filament given how unlikely people are to check any warning dialogues presented. For a one-off print that only just doesn't fit it may not seem like an issue, but as soon as your creation involves multiple parts any such scaling is going to cause more harm than good.

SnapJ1, 2023-12-14:
Maybe I wasn't clear enough??

I am not expecting cura to scale models.
I am expecting it to acceprt models, as they are within printing limits.

1st models Y had to be scaled from 197 to 195, before cura accepted it. (Y max 200)
2nd models Z had to be scaled from 200 to 199, before it was accepted. (Z max 200)

Scaling was done only for test purposes, as I wanted to find out, why those models were not accepted by Cura, even both were within printing limits. (300x200x200)

Both models were accepted by Luban, Snapmakers own slicer, as they are. 

Noticeable is, 1st models Z size was 200, but it was accepted???
2nd models Z had to be scaled???

Asterchades, 2023-12-14:
It's the use of the word "should" that changed the meaning you were getting across. Saying that something "should" be reduced implies that you want that action to be taken - not that you had to take that action to achieve a particular result.

Just through some experimentation, it would appear that Cura likes a very small buffer on some axes - sometimes. The same model (just a rectangular prism, created by repurposing a Support Blocker) on two different printers has yielded both a 0.1996mm and 0mm Z margin, and both a 0.05mm and 0.65005mm X/Y margin respectively being required for the model to be valid. Your particular setup and model could easily be different, and clearly the two models you've tested _are_ different to each other.

It's unlikely to affect all that many people, and it's unlikely to affect most of their models, but something isn't _quite_ right here right at the very limits of the build volume. I'll flag it for investigation - though I can't help but feel it may be assigned a low priority given the extremely limited impact.

GregValiant, 2026-10-25:
I'll go ahead and close this.

closed 2026-10-25 by GregValiant