Ultimaker/Cura #17640: Cura Unable to slice on my v400 Printers. The Anycubic Predators in Cura still work, just the v400 do net anymore.

yeAdmiral, 2023-12-14:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

V400 FLsun

### Name abnormal settings

No, i used a normal settings

### Describe model location

Center

### Describe your model

Since a couple of weeks i cant slice anymore. i Have sliced 100s of these models before.

### Add your .zip here ⬇️

[CSL847.zip](https://github.com/Ultimaker/Cura/files/13671074/CSL847.zip)

5axes, 2023-12-14:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

No issue for me : 
[image]

yeAdmiral, 2023-12-14:
[FV_CSL847.zip](https://github.com/Ultimaker/Cura/files/13671233/FV_CSL847.zip)

Yeah sorry about that. Here it is.

5axes, 2023-12-14:
Interresting ...  I cannot even load the project.  
[image]

Should be analysed by the devellopement Team.

yeAdmiral, 2023-12-14:
Yes, same here. What do i have to do for them to notice that?

5axes, 2023-12-14:
By an Ultimaker and call the Hotline :)

GregValiant, 2023-12-14:
#17550 is about project files causing Cura to crash.
I asked @MariMakes to take a look when I opened it.
It's been 5 days without a response so I assume they are busy doing Makerbot stuff.

GregValiant, 2026-10-25:
I'll go ahead and close this.

closed 2026-10-25 by GregValiant