Ultimaker/Cura #17644: Unwanted slicing artefacts generated using Connect Infill Lines

PythonesqueSpam, 2023-12-14:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

CTC i3 Pro B (Prusa Clone)

### Reproduction steps

Open model that uses exposed infill on assembly parts; in this case Head (Amame) Headphones V2.01 ear cups
Turn on "Connect Infill Lines" setting.
Slice Model.

### Actual results

[image]

### Expected results

[image]

...but only showing connected infill lines where infill is not exposed.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/13675386/cura.log)
[AA8_conesClosedBack_Cura.zip](https://github.com/Ultimaker/Cura/files/13675394/AA8_conesClosedBack_Cura.zip)

5axes, 2023-12-14:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

PythonesqueSpam, 2023-12-14:
Like the one in the zip file?

5axes, 2023-12-14:
No the zip file is just a 3mf format. No Printer information/parameter

5axes, 2023-12-14:
Edit 
I understand Now ...  As it is labeled as Bug I thing it was a bug it's more a Enhancement request.

PythonesqueSpam, 2023-12-14:
[AA8_conesClosedBack_Cura.zip](https://github.com/Ultimaker/Cura/files/13675908/AA8_conesClosedBack_Cura.zip)

PythonesqueSpam, 2023-12-14:
The setting causes filament to be deposited where there is no wall and effectively in mid-air, so I'm unclear why you suggest it is an enhancement.

PythonesqueSpam, 2023-12-14:
FYI - I have also tested this in Prusaslicer 2.7 with their "Infill Anchor" settings and it behaves exactly the same way so perhaps you are correct :D

5axes, 2023-12-14:
Because Connect Infill Lines is an option to connect infill lines ... it doesn't care if there is wall or not... Even if in the setting guide it's written "where the infill meets the inner wall or skin" it 's not because you have set the number of wall to 0 that there is not an inner wall for the slicer.

PythonesqueSpam, 2023-12-14:
Do you think it would be a useful (or practicable) enhancement to have the setting not deposit filament where a wall is not present?

GregValiant, 2023-12-15:
This has come up before in regards to exposed infill.  There are a couple of things I'll mention.

1.  It's always better to give the slicer the model you want rather than the model you wish you had and then ask the slicer to alter it into what you want.  Cura is a slicer, not a CAD package.  If you need that internal baffling done in a particular fashion I would suggest you design it in as part of the model.  The fact that exposing the gyroid infill is "almost, but not quite" what you want doesn't make it a bug.
2. _"...where a wall is not present?"_  Ah, but there are walls there.  The fact that the slicer has been asked not to print those walls doesn't change that fact.  The walls are features of the model Cura has been given.

I don't see a bug here so I'll remove the tag and close this.  You are welcome to propose this as a feature request as @5axes suggested and you can still post here.  It is sort of a niche situation so I don't know how much traction it will get with the Cura team.  I hope you understand.

closed 2023-12-15 by GregValiant