Ultimaker/Cura #17645: Slicing failed with any version past 5.4

Murmelman, 2023-12-14:
### Cura Version

5.5 and 5.6

### Operating System

Windows 11

### Printer

Flsun V400

### Describe your model

I can slice and print every model with 5.4... 

### Add your .zip here ⬇️

[Erde.zip](https://github.com/Ultimaker/Cura/files/13676586/Erde.zip)

This is the latest file I tried... Before I had the same issue with every file I tried unter cura 5.5. I also tried with a simple cube and it failed... 

I already tried:
- moving or turning the model or changing settings like linewidth etc... 
- deleting the printer and adding a new one
- the default settings that come with the profile
- reinstalling Cura
- loading a backup that definitely worked
- deleting the whole config-folder any other cura version and reinstalling cura 

My problem with 5.4 is that it often crashes on startup so I have to try 3-4 times to get in... Otherwise I would just stay at 5.4.

Greetings :)

5axes, 2023-12-14:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

Murmelman, 2023-12-16:
Sorry that I just posted the stl file... So here you have a cura file from another (simpler) project that also failed. 

Is it possible that the windows 11 issue only occurs with one of 4 different printers? Because I'm able to slice with my snapmaker artisan profile, my anycubic kobra profile and my snapmaker J1 profile with cura 5.5 and 5.6 at the same pc.

Greetings

[FV_Wandhaken für Basisplatte.zip](https://github.com/Ultimaker/Cura/files/13692569/FV_Wandhaken.fur.Basisplatte.zip)

5axes, 2023-12-16:
Could be an other issue similar to :  

As I cannot even open the project.
[image]

GregValiant, 2023-12-16:
@5axes I can't open that project either.
What I think I'm seeing is that many of the projects I can't open are trying to load nozzles but there aren't any "variant" nozzles for the subject printer in my installation of Cura 5.6.0.  It has happened a couple of times with materials as well.

Cura 5.5 consistently opens the projects without any problem but 5.6 will not.  In the case of that particular project file my install of 5.5 has the FLSun 0.4 nozzle file for the V400.  The 5.6.0 variants folder has no FLSun sub-folder so I'm missing the nozzle files.

I wonder if we are seeing a problem with the Conan thing (whatever it is) and the Cura installations aren't complete.  I'm certainly missing some files and it appears that other people have the same problem.

5axes, 2023-12-17:
Duplicate :  ?

https://github.com/Ultimaker/Cura/issues/17683
https://github.com/Ultimaker/Cura/issues/17645
https://github.com/Ultimaker/Cura/issues/17640

GregValiant, 2023-12-18:
@Murmelman - where did you get the printer definition file for your V400?  It's authored by "Joseph Jackson" and is 9517 lines long.
The Cura definition file I have in my installation is authored by "Guslain, Cyril" and is 45 lines long.

Murmelman, 2024-01-03:
@GregValiant 
You had the right Idea... My definition was from [here](https://github.com/phnxdrgn536/FLSun-v400) as the printer wasn't natively supported by cura. This File caused the problems. 

Thank you!

GregValiant, 2026-10-25:
I'll go ahead and close this.

closed 2026-10-25 by GregValiant