Ultimaker/Cura #17646: Tab + Enter uses incorrect value from text box

Coopski101, 2023-12-14:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

N/A

### Reproduction steps

1. Select item on prepare screen 
2. Ctrl+M to open multiply dialog
3. press tab to access number field
4. backspace, type '4'
5. press enter
6. item multiplied once

To confirm, did again 
1. When opened, dialog has '4' from last time it was open
2. tab, change back to '1' 
3. enter
4. item multiplied 4 times

### Actual results

item multiplied the number of times that were entered on the previous opening of the dialog, despite what number was entered

### Expected results

It should've multiplied 4, then 1 time respectively

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-12-15:
Thanks for the report.
That's a weird one.  The minimum value for the box is "1" so when I tried to get rid of the "1" by backspacing it didn't work because that would have resulted in a literal value of "0".  And that is how I ended up with 14 copies.
The only way I found to get rid of the "1" was to slide over it to select it, and then type in the "4".  I don't know if this is fixable considering the QT controls.  The Cura team will take a look.

GregValiant, 2026-10-25:
I'll go ahead and close this.

closed 2026-10-25 by GregValiant