Ultimaker/Cura #17648: GMC Truck Slice Failing

Tyman108, 2023-12-14:
### Cura Version

5.6

### Operating System

windows 11

### Printer

Anycubic Kobra

### Describe model location

center at 45deg -1.5mm

### Describe your model

gmc truckk

### Add your .zip here ⬇️

[AK_trruck v1_fixed.zip](https://github.com/Ultimaker/Cura/files/13679081/AK_trruck.v1_fixed.zip)

5axes, 2023-12-14:
An other Print Thin Walls issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452
[image]

Tyman108, 2023-12-27:
> An other Print Thin Walls issue : [#16440 (comment)](https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452) ? ![image](https://private-user-images.githubusercontent.com/11015345/290685365-ddd68da3-1e67-420b-8c4d-870695f771d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2ODkyMjksIm5iZiI6MTcwMzY4ODkyOSwicGF0aCI6Ii8xMTAxNTM0NS8yOTA2ODUzNjUtZGRkNjhkYTMtMWU2Ny00MjBiLThjNGQtODcwNjk1Zjc3MWQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI3VDE0NTUyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI4ZWUxZjEwYmRlOTM4OTU5OTJjZWE3ZWRhYTM3YTU5Mzk4YmQ1YjE5MjlkNjcxOWVkYjI0MTRkYTYwMjVlYTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ua4wv5TAcGYVRhK81EHAByyFmha4lkI4VnhpEpwvaNo)

how where you able to get this to slice?

closed 2024-04-03 by HellAholic