Ultimaker/Cura #17649: Slicing failed

staltux, 2023-12-15:
### Cura Version

5.5.0

### Operating System

Slackware Linux 15

### Printer

Elegoo Neptune 3

### Describe model location

rotating in Y axis solved

### Describe your model

craked D20

### Add your .zip here ⬇️

[cracked_D20_normal_scale.zip](https://github.com/Ultimaker/Cura/files/13679743/cracked_D20_normal_scale.zip)

5axes, 2023-12-15:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205
No issue for me 
[image]

GregValiant, 2026-10-25:
I'll go ahead and close this.  Much was fixed in Cura 5.7

closed 2026-10-25 by GregValiant