Ultimaker/Cura #17650: "sign" error causes head crash into build plate

zebrabunny, 2023-12-15:
### Cura Version

5.3.1

### Operating System

Win 11

### Printer

Anycubic Kobra Max V.1

### Reproduction steps

I am trying to print a "mouse head" costume shell.
I plated it in cura but it mostly wanted to crash, or print without proper build plate support so that the first layers of the actual piece were printed out in air.
I moved it 0.2mm into the build plate and that made it fail slicing completely.
Eventually I pulled it up a couple cm, and it sliced that way, but on the preview page, there were double images, and the print head behaved like the image (red headed one) would suggest, crashing into the build plate.

### Actual results

loud crashing noises

### Expected results

the expected results would be the expected results. a couple days of printing

### Add your .zip and screenshots here ⬇️

[AKMAX_KigAnthro1-Head.zip](https://github.com/Ultimaker/Cura/files/13680710/AKMAX_KigAnthro1-Head.zip)
 - Below: head pulled high -
[image]
- Below: results with high head (preview page)
[image]
- Below: head plated properly using automatic placement -
[image]
- Below: The unprintable output with floating arcs  (preview page) -
[image]
and of course, the log file
[cura.log](https://github.com/Ultimaker/Cura/files/13680904/cura.log)

GregValiant, 2023-12-15:
Thanks for the report.
Post that "Bad" gcode file.  I'm unable to duplicate the problem with your project file.
You have the Z-Offset plugin loaded.  In the project file it is set to "0".  Did you have it set to something else for the problem slice?

In regards to the preview "double image" that will happen when the model is raised above the build plate and "Remove Empty first Layers" is checked.  The model stays raised but the slice appears on the build plate.
There is a bug there though.  "Remove Empty First Layers" is not visible unless you turn off "Generate Support" first.

I notice you have Z-hops enabled at 0.02mm.  That's 1/3 of a human hair.  Can you really tell a difference with that?

[image]

zebrabunny, 2023-12-16:
tomorrow I'll send data. the 0.02 was about little crashy noises during print. seems to have worked, even tho it is a tiny amount  As the machine wears out the delrin wheels holding the base the build plate sinks a little, and the -0.02 makes a noticible difference there (getting hings to stick)

currently using cura to funnel gcode from slic3r but when the print is finished...

zebrabunny, 2023-12-16:
[BED_CRASG.zip](https://github.com/Ultimaker/Cura/files/13693718/BED_CRASG.zip)
there is the 1st section of the gcode file that crashed the head  (the zip was 70mb and refused to upload, if you need the whole thinf let me know and I'll rar it

GregValiant, 2023-12-16:
Thanks.
I've looked hundreds of Cura gcodes and I've never seen this.
;MINZ:**-31.49**
....
;LAYER:0
....
;TYPE:SKIRT
G1 F2400 **Z-31.49**
At the very end of the gcode file is a section where all the lines start with "SETTING_3".  If you could pass that section along maybe it will show something.

zebrabunny, 2023-12-16:
;Retract the filament
G92 E1
G1 E-10 F300
G28 X0 Y0
M84
M82 ;absolute extrusion mode
M104 S0
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = PLA adapt 150m\\
;SETTING_3 ndefinition = anycubic_kobra_max\\n\\n[metadata]\\ntype = quality_cha
;SETTING_3 nges\\nquality_type = pla\\nsetting_version = 21\\n\\n[values]\\nadap
;SETTING_3 tive_layer_height_enabled = True\\nadaptive_layer_height_variation = 
;SETTING_3 0.2\\narcwelder_allow_dynamic_precision = True\\narcwelder_allow_trav
;SETTING_3 el_arcs = True\\narcwelder_enable = False\\narcwelder_g90_influences_
;SETTING_3 extruder = True\\nbridge_settings_enabled = True\\nlayer_height = 0.3
;SETTING_3 \\nlayer_height_0 = 0.3\\nmaterial_bed_temperature = 40\\nmaterial_be
;SETTING_3 d_temperature_layer_0 = 60\\nsupport_enable = True\\nsupport_structur
;SETTING_3 e = tree\\n\\n", "extruder_quality": ["[general]\\nversion = 4\\nname
;SETTING_3  = PLA adapt 150m\\ndefinition = anycubic_kobra_max\\n\\n[metadata]\\
;SETTING_3 ntype = quality_changes\\nquality_type = pla\\nintent_category = defa
;SETTING_3 ult\\nposition = 0\\nsetting_version = 21\\n\\n[values]\\nbridge_skin
;SETTING_3 _support_threshold = 5\\nbrim_outside_only = False\\nbrim_replaces_su
;SETTING_3 pport = False\\ncool_fan_full_at_height = 0.6\\ncool_fan_speed = 100\
;SETTING_3 \ncool_fan_speed_max = 100\\ncool_fan_speed_min = 60\\ncool_min_tempe
;SETTING_3 rature = 210\\ninfill_pattern = cubic\\ninfill_randomize_start_locati
;SETTING_3 on = True\\ninset_direction = outside_in\\nmagic_mesh_surface_mode = 
;SETTING_3 normal\\nmaterial_final_print_temperature = 230\\nmaterial_initial_pr
;SETTING_3 int_temperature = 230\\nmaterial_print_temperature = 230\\nmaterial_p
;SETTING_3 rint_temperature_layer_0 = 230\\noptimize_wall_printing_order = False
;SETTING_3 \\nretraction_hop = 0.02\\nroofing_layer_count = 1\\nskirt_brim_speed
;SETTING_3  = 30\\nspeed_layer_0 = 90\\nspeed_print = 150\\nspeed_print_layer_0 
;SETTING_3 = 15\\nspeed_topbottom = 150\\nspeed_travel = 300\\nspeed_travel_laye
;SETTING_3 r_0 = 90\\nspeed_wall_0 = 150\\nspeed_wall_x = 150\\nspeed_z_hop = 40
;SETTING_3 \\nsupport_brim_width = 15\\nsupport_tree_max_diameter = 50\\nsupport
;SETTING_3 _wall_count = 2\\ntop_bottom_thickness = 2\\ntop_thickness = 2\\nwall
;SETTING_3 _overhang_angle = 60\\nwall_thickness = 2\\n\\n"]}

zebrabunny, 2023-12-19:
I did not have the z-offset plugin installed (from the marketplace)