Ultimaker/Cura #17655: Tree supports fail slicing

Error00418, 2023-12-15:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3

### Name abnormal settings

Tree supports caused it

### Describe your model

Yes, only without enabled tree supports, normal supports slice it fine

### Add your .zip here ⬇️

[CE3PRO_jogger_handle_8.1_top_v1.zip](https://github.com/Ultimaker/Cura/files/13684031/CE3PRO_jogger_handle_8.1_top_v1.zip)

5axes, 2023-12-15:
New Tree Support issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Slice Failed due to **Tree Support** 
https://github.com/Ultimaker/Cura/issues/17591
https://github.com/Ultimaker/Cura/issues/17597
https://github.com/Ultimaker/Cura/issues/17582
https://github.com/Ultimaker/Cura/issues/17578
https://github.com/Ultimaker/Cura/issues/17575
https://github.com/Ultimaker/Cura/issues/17549
https://github.com/Ultimaker/Cura/issues/17529
https://github.com/Ultimaker/Cura/issues/17521
https://github.com/Ultimaker/Cura/issues/17520
https://github.com/Ultimaker/Cura/issues/17479
https://github.com/Ultimaker/Cura/issues/17412
https://github.com/Ultimaker/Cura/issues/17351
https://github.com/Ultimaker/Cura/issues/17331

With Cura versions 5.5 and 5.6, if the settings do not allow tree-shaped supports to reach the supported area (branch angle, trunk diameter, space to create the support), the software generates an error and does not continue the slicing operation. Expecting result : No support if they cannot be generate but a model sliced.

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc