Ultimaker/Cura #17656: Cura 5.6 -Won't Launch

jtprouty, 2023-12-15:
### Cura Version

5.6.0

### Operating System

Window 11

### Printer

Kobra Max

### Reproduction steps

I installed 5.6.0 yesterday and it just brings up a blank screen when it launches. It also won't launch the previous version I had installed (4.12.0). I've uninstalled and reinstalled 5.6.0 several times with the same result.

### Actual results

No change

### Expected results

Launching the program.

### Add your .zip and screenshots here ⬇️

[image]

ranran2808, 2023-12-18:
I’m having the same issues, I’ve looked everywhere for results to fix this and I can’t find any, it is immensely frustrating

jtprouty, 2023-12-18:
Uninstalled 5.6.0 and installed 5.5.0 - same results.

ProfPun, 2024-02-21:
Have 
you tried running it as administrator?

jtprouty, 2024-02-22:
That worked but I shouldn't have to run it as administrator.
s

ProfPun, 2024-02-22:
Agreed, but it was the only thing I found to work. Sometimes , after running it once this way, then it would then open normally. Not sure if it is a Cura problem, Windows 10 or some sort of security issue? I am running Kaspersky.