Ultimaker/Cura #17657: LARGE PRINT SLICE FAIL

jward44553, 2023-12-15:
### Cura Version

5.5.0

### Operating System

Windows

### Printer

KOBRA NEO

### Name abnormal settings

overhang support angle 85*

### Describe model location

Center of plate, touching plate.

### Describe your model

Top of a house

### Add your .zip here ⬇️

[second floor - Copy.zip](https://github.com/Ultimaker/Cura/files/13688273/second.floor.-.Copy.zip)

GregValiant, 2023-12-15:
Thanks for the report.
Which version of Windows are you running?
You haven't given us much to work with here.  A project file and description of the problem would be a good place to start.

jward44553, 2023-12-15:
Hey Greg!

I had no idea what you might need, my apologies.

Windows Home 11 version 22H2
AMD Ryzen 7 5800H
16GM ram

I was using cura 5.5.0 (updated since and it failed at 5.6.0 as well.

Tried tweaking a ton of setting to bypass the bug, nothing worked
(resizing, moving on plate, etc).

GregValiant, 2023-12-15:
We've been collecting information regarding some sort of interaction between Windows 11 and Cura (that's why I asked for the version).  I'm running Windows 10 and I haven't really had any issues.  If there is a problem between the Operating System and Cura I wouldn't be the person to ask about it.
You might try going back to 5.2.2 or even 4.13.1.  If the problem is between Win11 and Cura there might not be a fix for a while.