Ultimaker/Cura #17659: Dino Bank wont Slice

inch4276, 2023-12-15:
### Cura Version

5.4

### Operating System

windows 11

### Printer

Anycubic Vyper

### Describe your model

no

### Add your .zip here ⬇️

[DinoBank_Solo.zip](https://github.com/Ultimaker/Cura/files/13689059/DinoBank_Solo.zip)

GregValiant, 2023-12-16:
Thanks for the report.
There are a growing number of reports that Windows 11 installations of Cura won't slice.  Data is being collected but at this point I have no idea if anyone is working on this.

@5axes here is another one for the list.

inch4276, 2023-12-16:
It's just this model. I have made and printed a ton of other things and Cura slices fine. But for some reason this Dino bank doesn't work.

5axes, 2023-12-16:
Ok actual list  : 
Slice Failed on Windows 11

https://github.com/Ultimaker/Cura/issues/17659
https://github.com/Ultimaker/Cura/issues/17645
https://github.com/Ultimaker/Cura/issues/17636
https://github.com/Ultimaker/Cura/issues/17635
https://github.com/Ultimaker/Cura/issues/17617
https://github.com/Ultimaker/Cura/issues/17613
https://github.com/Ultimaker/Cura/issues/17612
https://github.com/Ultimaker/Cura/issues/17595
https://github.com/Ultimaker/Cura/issues/17571
https://github.com/Ultimaker/Cura/issues/17567
https://github.com/Ultimaker/Cura/issues/17566
https://github.com/Ultimaker/Cura/issues/17552
https://github.com/Ultimaker/Cura/issues/17546
https://github.com/Ultimaker/Cura/issues/17538
https://github.com/Ultimaker/Cura/issues/17530
https://github.com/Ultimaker/Cura/issues/17526
https://github.com/Ultimaker/Cura/issues/17523
https://github.com/Ultimaker/Cura/issues/17509
https://github.com/Ultimaker/Cura/issues/17491
https://github.com/Ultimaker/Cura/issues/17490
https://github.com/Ultimaker/Cura/issues/17461 ??
https://github.com/Ultimaker/Cura/issues/17380
https://github.com/Ultimaker/Cura/issues/17343
https://github.com/Ultimaker/Cura/issues/17335
https://github.com/Ultimaker/Cura/issues/17253

GregValiant, 2023-12-16:
@inch4276 It took a while, but the bank sliced for me on Windows 10 Pro using Cura 5.6.0.  I really haven't any trouble slicing things with 5.6.0 (5.4 and to a lesser extent 5.5 had problems).