Ultimaker/Cura #17663: unsupported operand type(s) for +: 'float' and 'NoneType'

bigbm2019, 2023-12-15:
### Cura Version

5.5.0

### Operating System

macOS 14.2 (23C64)

### Printer

ender 3 with klipper

### Reproduction steps

I am trying to save the file 

### Actual results

it says unsupported operand type(s) for +: 'float' and 'NoneType'

### Expected results

it should save as gcode 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-12-16:
Thanks for the report.
Do you happen to have the "Z-Offset" plugin from the MarketPlace loaded?

GregValiant, 2023-12-16:
Thanks for the report.
This might be related to #17650 .  We need a project file to trouble shoot.  Use the "File | Save Project" command then zip the file and post the zip folder here.

closed 2023-12-30 by github-actions[bot]