Ultimaker/Cura #17667: Failed Slicing

calif94577, 2023-12-16:
### Cura Version

5.4.0

### Operating System

Mac OS 14.0

### Printer

SV-01

### Describe your model

Just this model

### Add your .zip here ⬇️

[SSL_Tabaxi Female Hexblade.3mf.zip](https://github.com/Ultimaker/Cura/files/13690981/SSL_Tabaxi.Female.Hexblade.3mf.zip)

5axes, 2023-12-16:
An other Print Thin Walls Issue ... https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

Model muste be print with support and at least a 0.2 Nozzle

calif94577, 2023-12-16:
I've definitely tried with support with no luck. However I need a nozzle smaller than or equal to 0.2 you're saying?

calif94577, 2023-12-19:
Just tried to slice it with supports and 0.2, 0.1, and even tried 0.05 nozzle without success. Same error.

closed 2024-04-03 by HellAholic