Ultimaker/Cura #17668: M106 for IDEX

silver341, 2023-12-16:
### Cura Version

5.4.0 and 5.6.0

### Operating System

windows 10

### Printer

IDEX

### Reproduction steps

Settings for fan of second hotend

### Actual results

Settings for second hotend not include in result gcode, only M106 S***, not M106 P* S***

### Expected results

fan of second hotend not work

### Add your .zip and screenshots here ⬇️

no

silver341, 2023-12-16:
---

GregValiant, 2023-12-16:
Thanks for the report.
If the fans have the same "number" then there will be no "P" parameter.  If they are numbered differently in the Machine Settings then the "P" gets added.  It does appear that it is not added correctly.  This is from a slice I just made:
T1
G92 E0
M105
M109 S225
M104 T0 S175
**M106 S255 P1**
At the next tool change:
T0
G92 E0
M105
M109 S195
M104 T1 S175
**M106 S255**
Without the "P" parameter most firmware would accept that as "active extruder fan".  What I don't see at all within the gcode is either fan ever being _turned off_ at tool change.  I consider that to be a bug.  It seems to have been present back to at least 4.13.1.  It's possible that turning on P1 would automatically turn off P0 but that would be done in the firmware and generally a printer/processor will do nothing without being told.  What I see in the gcode is that both fans would go to "regular fan speed" and unless there were some short layers to affect it, they would stay there till the end of the print.

IMO The gcode should be something like this when switching to T1:
G92 E0
M106 S0 P0
T1
M106 S76 P1
G92 E0
and then when switching to T0
G92 E0
M106 S0 P1
T0
M106 S76 P0
G92 E0

You will get that if you install this handy dandy 'beta' version of "AdvancedCoolingProfile".  Unzip the file and copy it to your configuration folder and the "scripts" folder.  Use the "Help | Show Configuration Folder" to find it.
Mine is in:  C:\Users\...my name...\AppData\Roaming\cura\5.6\scripts.
If there are problems (or you have comments) let me know.  I have been asked to make some changes so right now this is still a work-in-process.

[AddCoolingProfile.zip](https://github.com/Ultimaker/Cura/files/13692570/AddCoolingProfile.zip)

silver341, 2023-12-19:
Thank you very much, I missed that there is a cooling fan number in the settings