Ultimaker/Cura #17670: Program crash - problem with Tree Support

rstoki, 2023-12-16:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender-3 S1 Pro

### Name abnormal settings

Generate Support: True; Support Structure: Tree 

### Describe model location

The model is in the default location (in the center -- after Drag'n'Drop of STL file). The model is rotated 90 deg. counter-clockwise around green axis

### Describe your model

Just a simple hook for cables created in Blender. It contains small (2.5mm) holes (cubical)

### Add your .zip here ⬇️

[Cura 5.6 BUG-report.zip](https://github.com/Ultimaker/Cura/files/13693044/Cura.5.6.BUG-report.zip)

How to reproduce:
 - open Cura
 - Drag-and-Drop the attached STL file 
 - rotate the model 90 deg. counter-clockwise (around green axis)
 - turn "Generate Support": **ON**
 - select "Support Structure: **Tree**
 - Cura 'crashes' (probably an unhandled exception) 

If the model is sliced without Supports => it works 
If the model is sliced with Supports, and the Support Structure is set to "Normal", it works. 
If the model is sliced without rotation => it works. 

So the problem is probably with the Tree support structures, and maybe the combination of small holes.

5axes, 2023-12-16:
An other Tree Support and Part position issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc