Ultimaker/Cura #17671: Slicing failed with an unexpected error. [5.6.0]

Balcken, 2023-12-16:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

The same model has worked in previous versions

### Actual results

Nothing I have default settings

### Expected results

Slicing to work again

### Add your .zip and screenshots here ⬇️

[image]
[cura.zip](https://github.com/Ultimaker/Cura/files/13693396/cura.zip)

5axes, 2023-12-17:
An other "Print Thin Walls" Issue :https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-6793499

[image]

MariMakes, 2024-01-18:
Hi @Kiefels2,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes