Ultimaker/Cura #17672: Changed support to "Tree" and "Touching baseplate"

DouglasThom, 2023-12-16:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Snapmaker Artisan

### Name abnormal settings

Settings as defined by Snapmaker in their 0.9.4 plugin

### Describe model location

Model in default position

### Describe your model

Miniature with base plate separate and beside

### Add your .zip here ⬇️

[ec62f08518d64d789e16fca8f0abf340.zip](https://github.com/Ultimaker/Cura/files/13693631/ec62f08518d64d789e16fca8f0abf340.zip)

5axes, 2023-12-17:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

But it looks like a "classical" Windows 11 issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

No issue on Windows 10 : 
[image]

DouglasThom, 2023-12-17:
Ah! That's quite the Windows 11 issue list!

Thank you, @5axes , looks like I just switch back to Luban then and just be patient for the next version(s). I also appreciate you mentioning the support submission guide.

Take care!

-Doug

closed 2023-12-17 by DouglasThom