Ultimaker/Cura #17676: 2 Color Frog IDEX slicing failure

drewmoseley, 2023-12-16:
### Cura Version

5.6.0

### Operating System

Linux

### Printer

Sovol SV04 Dual Mode

### Describe model location

Bed Center

### Describe your model

2 Color Tree Frog

### Add your .zip here ⬇️

[2color_treefrog_small_solid.zip](https://github.com/Ultimaker/Cura/files/13694114/2color_treefrog_small_solid.zip)

5axes, 2023-12-17:
AN other "Print Thin Walls" Issue: https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

 .. on extruder 1 

[image]

MariMakes, 2024-01-18:
Hi @drewmoseley,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 8 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.

Hopefully, we can find a solution soon 🤞

HellAholic, 2024-02-16:
Small update,
Issue is resolved in CURA-11570 and will be part of the next Cura release.

closed 2024-02-16 by HellAholic