Ultimaker/Cura #17677: Linux Arm Support

castlec, 2023-12-17:
### Is your feature request related to a problem?

In the past, the community provided arm builds for Cura. Those are not becoming quite outdated. Linux X86 is supported through app image, as are Mac x86 and Mac Arm. With larger SBCs commonly being utilized as desktop devices, Cura on Linux Arm would close a gap in that ecosystem.

### Describe the solution you'd like

I'd like Linux/Arm added to the supported platforms.

### Describe alternatives you've considered

I've attempted using older builds. The newest build available is 5.3.0 in Fedora. I would be interested in helping to produce the new builds; however, I have not been able to get conan based builds to complete in x86 as I haven't been able to successfully reproduce the build environment.

### Affected users and/or printers

anyone using linux on arm. there are arm chromebooks and multiple SBC that are capable of running the software.

castlec, 2023-12-21:
Sharing this for the benefit of others.

I have now managed to run Cura 5.6.0 from source on jammy/armbian/aarch64/rk3588. It required the following.

- Build and install QT from source. I couldn't find a QT package that met requirements for 6.4.2 (cura 5.6.0) nor 6.6.0 (cura 5.7.0)
- Manually install PyQT6. This was particularly painful because the build automation is entirely broken for this package due to licensing. Because there is no binary distribution, this package must be built from source as well. The package, while compiling, requires license confirmation. `--config-settings --confirm-license --verbose` must be passed to `pip install PyQt6`. If this is not done, the build will hang forever. `pip install -r requirements.txt` does not accept the additional arguments. Install must be called on the package itself.
- Remove all dependencies declared on PyQt6 sub-packages. These don't exist since there is no binary package for PyQt6. The PyQt6 source build meets these dependencies. It seems like conan should be managing that knowledge. This must be done for Cura and Uranium (I feel like there's another package but I don't recall it right now).

I have yet to have success with app image packaging. There are multiple reasons for this. 
1. I have no idea what I'm doing
2. Cura code assumes x86_64 all over the place, even when architecture variables are being utilized.
3. Like Cura, AppImage itself has no aarch64 package. This means it must be used from another distribution mechanism. The best option for arm right now is `pip install appimage-builder`. It seems like using the python distribution rather than the `.AppImage` distribution will meet the needs of the system generically and relieve the CI of some of the funny embedded bashery.  Use `export APPIMAGEBUILDER_LOCATION` in environment to utilize different install location.
5. AppImage tries to ensure the newest dependencies are available. This breaks on armbian because it doesn't follow their required PEP 440. I'm not sure if this is their requirement or something that's coming in from stdlib but it's broken on armbian nonetheless due to the way armbian denotes origin of packages.  <upstreamversion><upstreamsource><armbianversion>. I had to open appimagebuilder/modules/deploy/apt/package.py to fux the version comparisons. A dumb replace of the string `ubuntu` made this work.
6. After these changes, I have a package built. It doesn't work though. See 1.

I was hoping to be able to post an appimage build for others to test but that's not going to happen for a while as it stands. Maybe tomorrow's adventures will bring something forward there.

Now that I understand enough of what's going on, I may try to find success with a cross compile. That seems like it's the best way to have arm packages in the existing github actions based CI system. Regardless, distribution needs to be figured out whether it exists as a fork or as a main Cura distribution.

makershopfi, 2024-06-20:
Hi!

Can you put some details about building cura?

- What Qt6 dependencies to remove and which files
- What Qt6 packages needs to installed manually
- Versions of conan+python/pip when building
- OS version

I have tried several times to build cura but failed to get it running. I would like to use 5+ version as now I'm using 4.13...

miikasyvanen, 2024-06-20:
Hi!

Can you put some details about building cura?

- What Qt6 dependencies to remove and which files
- What Qt6 packages needs to installed manually
- Versions of conan+python/pip when building
- OS version

I have tried several times to build cura but failed to get it running. I would like to use 5+ version as now I'm using 4.13...

Thanks,
Miika

castlec, 2024-06-20:
I'm away from home for the next three weeks so I can't go pull more information for you. 

Here's what I can tell you:
* Use the qt and pyqt versions as specified in the branch you're using. Download from qt whatever version is specified in the Cura builds then build and install. It's not necessary to build all packages but building them all is the easiest (but slowest) way to ensure you have everything. 
* I was following the local run instructions for Cura. Whatever it says to use for Conan and pip are what you should use. 
* You'll want to tell Conan the specific version you're trying to build when you execute. I recall querying their remote to see what versions had specified dependencies and building one of those. This isn't a necessity but will help you since you won't know if there are bugs in main.
* Removing the extra pyqt dependencies during the build is pretty easy. You literally open up requirements.txt and remove all packages that are not the base pyqt package. I believe the base package is pyqt-qt but it may be pyqt. If it's not one of those two, remove it. It may be necessary to remove pyqt-qt also. If pip can't find it, remove it. It'll be in your package built by pip. This must be done for all Cura packages built by Conan. The easiest thing to do here is wait until the build fails then modify the requirements.txt for that package, pip install using the same python env Conan is using for that package, then kick off the Conan build again. 
* I built on Armbian Ubuntu/jammy, which I believe matched the Cura build system.
* If there's a matching pyqt package for your distribution, you can skip the manual builds but you'll still need to remove the hash codes from requirements.txt or figure out how to override the specified version otherwise.

I think that's all of it. 

If you want to post up a build log, I can have a look at it.

makershopfi, 2024-06-21:
Thanks for the quick reply!

I think my biggest problem is installing Qt. I have managed to build Cura but when I run it, it complains about missing Qt packages. Do you remember from where you downloaded the sources and how did you exactly build them? Should it done in venv or just system wide etc.?

castlec, 2024-06-21:
Find your dependency version here. https://download.qt.io/archive/qt/

Be sure to read about configuration options. That will help you ensure you're not missing packages.
https://doc.qt.io/qt-6/configure-options.html

miikasyvanen, 2024-06-27:
### Building Cura

Ok, today managed to build Cura 5.7.0 on OrangePi 5 Pro 16GB board running Ubuntu Jammy (OrangePi OS version). Here are the steps that I did, some might be unnecessary but these should get build complete and Cura running. In this build Cura git clone is made in ~/Development so build dir is ~/Development/Cura.

#### Python and pip version used in build
- Python 3.10.12
- Pip 24.1.1 (default was 22.x, need to upgrade after first failed build)

#### Install Qt6 binaries with online installer. I used Qt 6.7.0 in this install.
https://www.qt.io/download-qt-installer-oss

At least under additional libraries all should be checked.

#### Install and configure Conan
```
pip install conan==1.64
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
```

In .conan/conan.conf change level-option in [log]-section, this helps to see when build will wait for license input. If not stopped, it stays there waiting forever and consumes all RAM and in the end, your machines is not responsive anymore.
```
[log]
run_to_output = True        # environment CONAN_LOG_RUN_TO_OUTPUT
run_to_file = False         # environment CONAN_LOG_RUN_TO_FILE
level = 10                # environment CONAN_LOGGING_LEVEL, default is warn
print_run_commands = True   # environment CONAN_PRINT_RUN_COMMANDS
```

#### Clone Cura
```
git clone https://github.com/Ultimaker/Cura.git
cd Cura
```

#### Add Qt install path to PATH and set makeflags to compile with all processors to speed up build (optional)
```
export PATH=/opt/Qt/6.7.0/gcc_arm64/bin:/opt/Qt/6.7.0/gcc_arm64/lib:$PATH
export MAKEFLAGS="-j$(nproc)"
```

#### Initialize the Virtual Python Development Environment
```
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
```

Once build hangs on PyQt6 license checking, stop building with 'Ctrl+C'

#### Upgrade Pip
```
- ~/Development/Cura/venv/bin/python -m pip install --upgrade pip
```

#### Install PyQt6

First, remove/comment out all PyQt6 modules except PyQt6 itself and PyQt6-sip from Cura and Uranium requirements.txt files
```
~/Development/Cura/requirements.txt
find ~/.conan/ -name requirements.txt # I have this at path ~/.conan/data/uranium/5.7.0/_/_/package/e541e632dd3ca870d37dab822cdf6eaa3df15dca/pip_requirements/requirements.txt
```

#### Change PyQt6==6.6.0 to PyQt6==6.7.0 and change in both files hash to
```
    --hash=sha256:3d31b2c59dc378ee26e16586d9469842483588142fc377280aad22aaf2fa6235
```

#### Remove possible Qt files from venv
```
sudo rm -r venv/lib/python3.10/site-packages/PyQt6*
```

#### Purge pip cache
```
~/Development/Cura/venv/bin/pip cache purge
```

#### Install PyQt6==6.7.0
```
~/Development/Cura/venv/bin/pip install PyQt6==6.7.0 --force-reinstall --config-settings --confirm-license= --verbose
```

#### Restart conan build
```
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
```

#### Activate the Virtual Python Environment
```
source venv/bin/activate
```

#### Run Cura
```
python cura_app.py
```

miikasyvanen, 2024-06-29:
### Building Cura AppImage after building Cura itself using above instructions

Tested with:
- Orange Pi 5 Pro
- Orange Pi Ubuntu Jammy

#### Build Ultimaker-Cura folder with all necessary files for standalone use
```
cd ~/Development/Cura
pyinstaller venv/conan/UltiMaker-Cura.spec
```
If you get error message about Qt5 you need to exclude it in 'Analysis' section in the end of UltiMaker-Cura.spec-file
```
excludes=['PyQt5'],
```

#### Download appimagetool for aarch64
```
cd dist/
wget https://github.com/AppImage/appimagetool/releases/download/continuous/appimagetool-aarch64.AppImage
```

#### Copy cura icon
```
cp ~/Development/Cura/resources/images/cura-icon.png UltiMaker-Cura/
```

#### Create desktop entry
```
nano UltiMaker-Cura/cura.desktop
```

```
[Desktop Entry]
Name=UltiMaker Cura
Name[de]=UltiMaker Cura
GenericName=3D Printing Software
GenericName[de]=3D-Druck-Software
GenericName[nl]=3D-Print Software
Comment=Cura converts 3D models into paths for a 3D printer. It prepares your print for maximum accuracy, minimum printing time and good reliability with many extra features that make your print come out great.
Exec=UltiMaker-Cura %F
Icon=cura-icon
Terminal=false
Type=Application
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;image/bmp;image/gif;image/jpeg;image/png;text/x-gcode;application/x-amf;application/x-ply;application/x-ctm;model/vnd.collada+xml;model/gltf-binary;model/gltf+json;model/vnd.collada+xml+zip;
Categories=Graphics;
Keywords=3D;Printing;
X-AppImage-Version=5.7.0
Path=
StartupNotify=false
```

#### Create AppRun link to point UltiMaker-Cura executable
ln -sf UltiMaker-Cura/UltiMaker-Cura UltiMaker-Cura/AppRun

#### Build AppImage
```
ARCH=aarch64 ./appimagetool-aarch64.AppImage UltiMaker-Cura
```

That's it! Now you can copy/move AppImage to somewhere and use .desktop file to launch AppImage!
#### Don't forget to change line 'Exec=UltiMaker-Cura %F' in .desktop file to point to your AppImage!

ccgarant, 2024-12-31:
+1 on this Issue, would be great to have a linux aarch64 build. 

I am using a Mac Mini M1, ARM( aarch64), rooted for [Linux  Asahi Ubuntu](https://ubuntuasahi.org/)

Willing to test and support as needed.

miikasyvanen, 2024-12-31:
I have AppImage on releases for 5.8.0 version, if you want to check it out. Might update it to newest version when I have some free time.

https://github.com/miikasyvanen/ultimaker-cura-arm64