Ultimaker/Cura #17679: Lithophane files not slicing

Nikky2010, 2023-12-17:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

CrealityS1 Plus

### Name abnormal settings

Wall 999 Infill 0%

### Describe model location

It will slice if it is vertical but not once I rotate it to lay it flat

### Describe your model

I have sliced successfully multiple time this model & also the same model contaning different picture. Yestersay & today I have tried 4 or 5 varients of this model & older versions of cura & none of then will slice 

### Add your .zip here ⬇️

[Lithophane not slicing.zip](https://github.com/Ultimaker/Cura/files/13696056/Lithophane.not.slicing.zip)

5axes, 2023-12-17:
From now on, this can be considered a classic bug **Print Thin Walls**  : 

[image]

Note a Lithophane is normally printed vertically

Slice Failed due to **Print Thin Walls**

https://github.com/Ultimaker/Cura/issues/17679
https://github.com/Ultimaker/Cura/issues/17671
https://github.com/Ultimaker/Cura/issues/17606
https://github.com/Ultimaker/Cura/issues/17560
https://github.com/Ultimaker/Cura/issues/17528
https://github.com/Ultimaker/Cura/issues/17503
https://github.com/Ultimaker/Cura/issues/17499
https://github.com/Ultimaker/Cura/issues/17470
https://github.com/Ultimaker/Cura/issues/17467
https://github.com/Ultimaker/Cura/issues/17457
https://github.com/Ultimaker/Cura/issues/17456
https://github.com/Ultimaker/Cura/issues/17496
https://github.com/Ultimaker/Cura/issues/17478
https://github.com/Ultimaker/Cura/issues/17439
https://github.com/Ultimaker/Cura/issues/17438
https://github.com/Ultimaker/Cura/issues/17410
https://github.com/Ultimaker/Cura/issues/17367
https://github.com/Ultimaker/Cura/issues/17344
https://github.com/Ultimaker/Cura/issues/17177
https://github.com/Ultimaker/Cura/issues/16688
https://github.com/Ultimaker/Cura/issues/16515
https://github.com/Ultimaker/Cura/issues/16417

MariMakes, 2024-01-18:
Hi @Nikky2010,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 5 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞