Ultimaker/Cura #17681: Strange Infill path on the Top Surface

5axes, 2023-12-17:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3

### Reproduction steps

Project file included to the report.

### Actual results

Strange infill path on the Top Surface.
[image]

### Expected results

No infill path on the Top Surface

### Add your .zip and screenshots here ⬇️

[Protege-Prise-Noel-CouvercleV2.zip](https://github.com/Ultimaker/Cura/files/13696218/Protege-Prise-Noel-CouvercleV2.zip)

GregValiant, 2023-12-18:
Thanks for the report.
I can duplicate it.  I can't figure out what is causing that.  It's on multiple layers, but not around the entire print.
[image]

The workaround is to use "Connect Infill Lines".

The Cura team will take a look.

5axes, 2023-12-18:
The Cura team will take a look... or not ;)