Ultimaker/Cura #17683: Falied slicing a merged object

Kronologic81, 2023-12-17:
### Cura Version

5.6.0

### Operating System

Windows 11 x64

### Printer

FLSUN V400

### Name abnormal settings

Used the standard configuration of Normal Engineering. 0.2

### Describe your model

There are a Shelf supports downloaded from printedparts.com also i've ccorrected the mesh using the addons

### Add your .zip here ⬇️

[80kg_shelf Generic PLA lw 0.44mm lh 0.2mm if 10.0% ext1 210C bed 60C.zip](https://github.com/Ultimaker/Cura/files/13696552/80kg_shelf.Generic.PLA.lw.0.44mm.lh.0.2mm.if.10.0.ext1.210C.bed.60C.zip)

5axes, 2023-12-17:
an other issue with FLSUN V400

[image]

@GregValiant  looks like the same  problem as the report  : https://github.com/Ultimaker/Cura/issues/17640

Could be an issue regarding FLSUN V400 printer configuration on Cura 5.6.0 ?

GregValiant, 2023-12-17:
@5axes I commented in #17645 and I'm leaning towards it being a Cura issue.  My installation doesn't seem to have all the files present for the FLSun V400.  It isn't the only printer I've noticed that about.  The definition file within one of these bug reports was completely different than the definition file I have in my installation - they weren't even by the same author.
Here are the two "def.json" files.  The one from the project file is named as such and the other is from my Cura installation.
Within the project file it states that it's made with 5.6.0.  Something is very odd.
[flsun_v400 defnitions.def.zip](https://github.com/Ultimaker/Cura/files/13697941/flsun_v400.defnitions.def.zip)

5axes, 2023-12-18:
Semble correcte avec la définition FLSUN V400 de la version Cura.
[image]

GregValiant, 2023-12-18:
Worked for me so long as I didn't open it as a project.

@Kronologic81 The FLsun V400 definition file that is included with the project appears to be a non-Cura definition file.  "Joseph Jackson" appears as the author and the file is 9500+ lines long.  The Cura definition file was authored by "Cyril Guslain" and is only 45 lines long.
Another thing is that the profile "Normal Engineering. 0.2" sounds like it's an Ultimaker profile and shouldn't be available to other printers.
So this may well be a problem with the definition file you are using.