Ultimaker/Cura #17688: Slicing Skadi cover failed

PeterAlcivar, 2023-12-17:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Creality Ender-3 Max

### Describe model location

No

### Describe your model

This is a cup cover with name. I have printed the same thing before, the only difference being I increased the disk thickness from 2mm to 3mm.

### Add your .zip here ⬇️

[CE3MAX_Cup_Cover_Large_3_Skadi.zip](https://github.com/Ultimaker/Cura/files/13697095/CE3MAX_Cup_Cover_Large_3_Skadi.zip)

5axes, 2023-12-17:
You don't need Support for this part. And certainly not a Tree Support.

[image]

An other Tree Support issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Slice Failed due to **Tree Support** 

https://github.com/Ultimaker/Cura/issues/17688
https://github.com/Ultimaker/Cura/issues/17670
https://github.com/Ultimaker/Cura/issues/17655
https://github.com/Ultimaker/Cura/issues/17591
https://github.com/Ultimaker/Cura/issues/17597
https://github.com/Ultimaker/Cura/issues/17582
https://github.com/Ultimaker/Cura/issues/17578
https://github.com/Ultimaker/Cura/issues/17575
https://github.com/Ultimaker/Cura/issues/17549
https://github.com/Ultimaker/Cura/issues/17529
https://github.com/Ultimaker/Cura/issues/17521
https://github.com/Ultimaker/Cura/issues/17520
https://github.com/Ultimaker/Cura/issues/17479
https://github.com/Ultimaker/Cura/issues/17412
https://github.com/Ultimaker/Cura/issues/17351
https://github.com/Ultimaker/Cura/issues/17331

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm/
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc