Ultimaker/Cura #17689: Pause at height (Layer number) isn't working with Print Sequence One at a Time when higher than tallest object

peterkng, 2023-12-17:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Ender 3 Max

### Reproduction steps

1) Add two objects print
2) Set sequence to print one at a time
3) Slice and find what layers belong to second object
Example: Object One (10 layers tall) = layers 1 through 10 & Object Two (20 layers tall) = layers 11 through 30. Cura's Preview shows layers 1-10 for Object One and then layers 11-30 for Object Two

4) Add pause at height using a layer number belonging to second object which is also higher than the tallest object. 
Example: Object One (10 layers tall) = layers 1 through 10 & Object Two (20 layers tall) = layers 11 through 30. Add pause of height of 21 through 30

### Actual results

The gcode for pause is not included, the machine will not pause at height for the second object.

### Expected results

The machine should pause at height for the second object because Cura reports those layer numbers while using Preview.

### Add your .zip and screenshots here ⬇️

Note: Cura's Pause at Height seems to max out at the layer max height when printing All at Once

I have two objects, each 10 layers high and I use Print Sequence One at a Time. 
I slice and see that layers 1-10 belong to Object One and Layers 11-20 belong to Object Two.

I add Pause at Height at layer 9 and the M0 command shows up in GCode.
I add Pause at Height at layer 15 and the M0 command does not show up in Gcode.
[Pause at height with print sequence.zip](https://github.com/Ultimaker/Cura/files/13697520/Pause.at.height.with.print.sequence.zip)

GregValiant, 2023-12-18:
Thanks for the report.
"One at a Time" is a special situation.  Although the layers are numbered "0 to top layer" in Cura, they are numbered "per model" in the gcode.  That means each model starts on layer 0.  If you have 3 models that are 10 layers high then the gcode will show 0-9, 0-9, 0-9 and the Cura preview will show 1-30 layers.  As a consequence, if you try to pause at layer 20 it will be ignored because there isn't one in the gcode.
To fix this requires a re-write of Pause at Height but it may not be possible without re-numbering the gcode from "One-at-a-Time" numbering to "All-at-Once" numbering.

The Cura team will take a look.  Right now you can only pause within the first model if you are using "One at a Time".

GregValiant, 2025-01-15:
@peterkng 
This is my re-write of PauseAtHeight.  There are added options including the ability to pause in each part in "one -at-a-time" mode.
You can unzip the file and place the .py file into your "scripts" folder in the Configuration Folder.  It will be available under "Extensions" with the other post-processors.
[PauseAtLayer.zip](https://github.com/user-attachments/files/18423334/PauseAtLayer.zip)
You can open the file in any text editor.  At the top is the list of changes I made.
I have not yet requested that this be added to Cura as it needs more eyeballs to insure it works OK.
Let me know how it works for you.