Ultimaker/Cura #17692: [5.6] Slicer adds additional lines

soxhi8, 2023-12-17:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Slice model `Lid RC3 - 0.1mm Tolerance.stl` in Cura 5.6.0, with the settings that are attached.
2. Observe weird lines being added on layer 3.

[Lid RC3 - 0.1mm Tolerance.stl.zip](https://github.com/Ultimaker/Cura/files/13698069/Lid.RC3.-.0.1mm.Tolerance.stl.zip)
[Settings.zip](https://github.com/Ultimaker/Cura/files/13698070/Settings.zip)

### Actual results

[image]

### Expected results

[image]

### Add your .zip and screenshots here ⬇️

[3mf.zip](https://github.com/Ultimaker/Cura/files/13698076/3mf.zip)
[image]
[Settings.zip](https://github.com/Ultimaker/Cura/files/13698079/Settings.zip)
[Lid RC3 - 0.1mm Tolerance.stl.zip](https://github.com/Ultimaker/Cura/files/13698080/Lid.RC3.-.0.1mm.Tolerance.stl.zip)
[image]

GregValiant, 2023-12-18:
Thanks for the report.
I measured the base at 0.8 thick, and your layer height is 0.32 so it doesn't go into it evenly.  I think what's happening is that layer 3 is incomplete except for areas where Cura has adjusted the line width to fit.  In those areas you get the complete third layer but in others Cura doesn't want to go to 0.96 so there is no third layer.

To fix it, make the Initial Layer Height and the Layer Height 0.266 (which is about 0.8/3 ).  That way Cura won't be confused about that third layer.  The problem also goes away if you turn on Adaptive Layers
[image]

The Cura team will take a look.

soxhi8, 2023-12-18:
Thank you for the explanation and taking a look. Just to illustrate my use case: For the specific model, having a clean surface is my priority, so snapping up or down in total height would both work. The target for the print is speed, hence the layer height.
But I will keep this in mind for the moment while using Cura 5 and adjust accordingly.

soxhi8, 2023-12-19:
> The problem also goes away if you turn on Adaptive Layers

Today one of my models exhibits a similar issue while using adaptive layers. Changing layer height to 0.2mm and keeping adaptive layers on mitigates the issue.
Actual result:
[image]
Expected result:
[image]
The fun part is, that I first tried fixing my model, resulting in three different versions of the model. They all exhibit different variations of the error:
[image]
[Settings&3MF_2.zip](https://github.com/Ultimaker/Cura/files/13716619/Settings.3MF_2.zip)
[STL_2.zip](https://github.com/Ultimaker/Cura/files/13716620/STL_2.zip)

GregValiant, 2023-12-19:
Those three models look more like the top surfaces aren't exactly flat.

soxhi8, 2023-12-20:
Nevertheless, the way Cura deals with it seems strange to me.